niklasekstrom/a314 #15: The repo shouldn't contained compiled binaries; instead make releases

niklasekstrom, 2019-06-11:
Currently the compiled binaries are included in the Git repository. It turns out it's easy to forget to compile one binaries after a change, and then the source and binaries are out of sync; this has already happened and led to unnecessary trouble shooting. Probably remove the binaries from the repo and make proper releases (that can be hosted on GitHub as well.)

terriblefire, 2019-08-16:
My suggestion would be a top level makefile that traverses the structure and produces all the binaries in a new top level folder. 

Make clean would wipe all the binaries in the repo. 

You could probably even get a Quartus docker container and build the firmware at the same time. I’d have to look into that latter bit. Container might be too big to be practical

niklasekstrom, 2019-08-16:
"My suggestion would be a top level makefile that traverses the structure and produces all the binaries in a new top level folder."

This sounds like a good improvement. I tried to do something like this but wasn't able to have make automatically create the sub-folders in the output folder corresponding to the sub-folders in the source folder. Probably creating the needed sub-folders explicitly in the Makefile should be good enough.

I was also thinking about setting it up so that files can be compiled on a raspberry pi, as some people may use Windows laptops, but everyone with an interest in this project has at least one RPi.

niklasekstrom, 2023-08-13:
This was fixed at some point.

closed 2023-08-13 by niklasekstrom