niklasekstrom/a314 #16: Add power buffer to allow for proper RPi shutdown

Eriond, 2019-06-11:
Add a power buffer to the A314 5V power line, possibly in the form of a supercap, in order to maintain a short-lived supply to the RPi after the Amiga has been powered off. Also connect a "power sense" to an unused GPIO pin, to trigger proper shutdown procedure on the RPi.

Eriond, 2019-09-10:
This enhancement did not make into the 1.1 update. It is still pending for now; there is no committed time-frame.

arnljot, 2020-01-23:
This can be done in software, no need for any special hardware. Many open souce raspi based projects have solved this. OctoPi, and also MISTER Fpga for instance is running of a read only system partition, and has the read write partitions as memdisks. If the user wants a persistant r/w partition on the SD card, it can be provided, but just as with a real amiga hd, he'd be warned not to power on without letting write operations "settle in". Could even send an "unmount" command before power off.

I'd rather suggest that effort is spent by raspbian able techs so make a A314 distro that can download updates from this github, and mostly only be mounted in r/w mode for online updates.

Normal operation r/w is either in a memdisk or a user-partition that can be recovered without risking loosing system partition integrity. The boot partition should be read only, given the hw limitations of the raspi which is shares with it's host, the amiga.

closed 2024-05-21 by niklasekstrom