niklasekstrom/a314 #18: A600 support

tangorri, 2019-07-13:
Would be nice to add support for Amiga 600, or if it is allready supported, validate it by test and clarify the support in documentation and description of the project.

Eriond, 2019-07-13:
We're working on a A600 version of the hardware, and while it's still in the design phase, we can already now conclude it will only be able to support the Pi Zero form factor due to mechanical restrictions. More info will be posted once there is some real hardware produced. Probable time frame is Q1 2020.

Eriond, 2019-11-09:
**Update:** A600 version is now actively being worked on. We can already now see that there will be a couple of differences compared to the A500 version:
- The memory will be increased to 2MB, thus allowing for all chip memory to reside on the A314. (prices for 2MB SRAM chips have dropped recently to apx the same levels as 1MB had previously)
- Our intention is that programming (and re-programming) of the FPGA flash is made possible directly from the Raspberry Pi, without any additional hardware. Those who wish to continue using a classic JTAG interface can still do so, but will have to make a custom adaptor cable.

plasticuser, 2020-03-01:
If the A600 version can only support a Pi Zero due to physical limitations, could it alternatively support a CM3 compute module?

Eriond, 2020-03-01:
It could, but it won't because we see it as a too big re-design for very little, if any, advantages. So the cost/performance ratio is not justifiable.

plasticuser, 2020-03-02:
The cost is exactly the same as being able to use a full Pi 3 B+. It's more than a Pi Zero, but then the design brief calls using a Pi Zero a "design limitation"... ;)

The physical advantages are numerous.  Being able to place the connectors exactly where they are best suited. Being able to place only the connectors required. Lower power footprint by not using devices that are not installed. Extremely compact form.

I'd be happy to design this PCB.

niklasekstrom, 2020-03-02:
@plasticuser If you feel up to designing and building a version of A314 for A600 using a CM3 compute module then of course that would be a very positive development :)

My initial thoughts about such a design is: how will you add back networking (wifi) and USB capabilities that are not available in CM3 (as compared to in Pi Zero)? And what is the combined cost of CM3 + additional chips? If the cost is not a lot more than a Pi3 then this sounds very interesting.

plasticuser, 2020-03-03:
I recognize it would cost a "little" more. I'm estimating about $7-$9. 

The fun bit is that I could supply the board with build time configuration options to lower the price point. Happy with ethernet? Cool! Want lots of I2C or SPI break outs on the board? Cool! 2MB SRAM? Yisss!

It is worth $10 on the final product price to have the four cores, and the option of SSD or eMMC - which is MUCH faster and more reliable. The CM4 is going to have a different pinout, but it would be easy to adapt the design at that point for the new connector.

So, yes, I'll happily do it, and I'd prefer to do it within your project than forking it and creating confusion and lots of extra work for myself ;)

niklasekstrom, 2020-03-03:
@plasticuser That sounds great! Let us know if you have any questions or need help with anything.

plasticuser, 2020-03-04:
I shall.

niklasekstrom, 2020-08-21:
Fixed by commit c04a5190c468d0750ad19860da7e57ba250490ec.

closed 2020-08-21 by niklasekstrom