niklasekstrom/a314 #19: Raspian usecase?

apocalypse2012, 2019-07-13:
Hello,
Thanks for this innovative contribution.
I wanted to ask about use-cases for the a314. I am wondering if this device would make it possible to use the Amiga 500 as a peripheral dock for an Amibian based Raspberry Pi? Either using the Pi’s HDMI output or the a500s graphics via the shared graphics memory? Either way, using all the Amigas IO devices (floppy, mouse, ports, etc) as devices exposed to Amibian through the a314 device. This would allow a great old school user experience while enabling a much broader and more modern set of functionality that emulation provides.
Is this part of your concept or otherwise on the roadmap?

Raymond

niklasekstrom, 2019-07-14:
Hi Raymond,

Thanks for sharing your idea! It's not something I had considered before.

The way we have been thinking about A314 is that the RPi becomes a co-processor to the Amiga. It seems that what you are proposing is turning that assumption upside down, and instead the A500 (or at least the input/output devices of the A500) are under the control of the RPi.

I guess what you are proposing should be doable for some motivated individual, but at the same time it seems non-trivial. I don't think it's something that me or Eriond is going to pursue, as none of us is using Amibian. If you or anyone else wants to pursue this idea I'll be happy to answer any questions about A314!

closed 2019-07-14 by niklasekstrom

apocalypse2012, 2019-07-14:
I would be interested in pursuing this, but it would be slow going in the near term with my current load. I am a developer with some programming experience. My question is where to start. It seems you already have a base driver, and building a higher level protocols and driver APIs should be possible, possibly trivial for the most basic features. Can you give me any guidance on how to proceed from a hardware perspective? How are you provisioning dev hardware for yourselves?
I have a 500 and an rpi3 I can dedicate to this right now, but my hardware and soldering skills are not up to board fabrication.

Eriond, 2019-07-15:
As mentioned in the EAB A314 thread, there is a "program" where we are giving away hardware to motivated programmers with bold/interesting ideas, but that lack the skill to assemble the boards themselves. You may fall into that category :)
However, the supply of free boards is extremely limited, so we would like to see some sort of proof that you may be up to the task you're pursuing. One way to go would be to read up on the currently available source, and tell us your strategy on how you intend to proceed. Asking relevant programming-related questions, maybe forking the project, etc. will help as well.
At best, we could send you a board somewhere mid-August (since I'm on vacation right now).