niklasekstrom/a314 #20: Create a sound device in Linux (running on the RPi) that outputs sound via Paula

niklasekstrom, 2019-07-14:
I'm not sure how hard this would be, but if possible it could be nice to make a sound device for Linux, running on the RPi, that writes the sound samples to A314 memory and then sets it up on the Amiga side so that the sound is played by Paula. I'm guessing this would allocate two of the four Amiga audio channels.

aiobofh, 2019-07-15:
This would be really cool. Kind of a channel-mixer outputting 14-bit sound!? :)

niklasekstrom, 2019-07-16:
It seems like it may be possible to use the ALSA file (or shm) plugin (https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html) to obtain samples in a format suitable to write to Amiga chip memory.

This is what I've got so far for .asoundrc:

pcm.tofile {
    type file
    format raw
    file /tmp/out.raw
    slave.pcm null
}

pcm.amiga {
    type plug
    slave {
        pcm tofile
        format S8
        channels 2
        rate 16000
    }
    hint {
        description "Changes format to match Amiga"
    }
}

commit 947ce27c (2019-07-18): Added ability to play audio from RPi with Amiga as audio output device. You need to first run piaudio on the Amiga, which connects to piaudio.py that is running on the RPi. Then you can do 'pi mpg123 -a amiga file.mp3' to play file.mp3 and route the samples to the Amiga. Fixes #20.
closed 2019-07-18 by niklasekstrom