niklasekstrom/a314 #22: Implement networking through bsdsocket.library

niklasekstrom, 2019-07-25:
Write a bsdsocket.library that forwards network calls to the RPi.

The API seems straight-forward enough: https://wiki.amigaos.net/amiga/autodocs/bsdsocket.doc.txt.

Each time bsdsocket.library is opened a new Library structure should be allocated, initialized and returned, in accordance with the description of bsdsocket.library/OpenLibrary in the linked document.

niklasekstrom, 2019-08-08:
Just an update: I have made a fair amount of progress on this, and with a few more hours of hacking I should have a useful subset of the API functions implemented.

I was able to get amitcp_telnet (http://aminet.net/package/comm/tcp/amitcp_telnet) to run under WB 1.3 (there was a single 2.0 function called), so that's what I'm using to test with.

niklasekstrom, 2019-08-10:
I have implemented the following functions: socket, connect, send, recv, shutdown, setsockopt, IoctlSocket, CloseSocket, WaitSelect, SetErrnoPtr, Inet_NtoA, inet_addr, gethostbyname, getservbyname. These are all the functions that amitcp_telnet call, and so that program now works.

The trickiest part has been to realize which arguments need to be byte-swapped due to that the Amiga is big-endian and the RPi is little-endian.

Implementing the rest of the functions should be conceptually straight forward but a bit time consuming. If anyone wants to use this functionality then let me know and I will put in some effort to finish it up.

nonarkitten, 2019-08-16:
I think this is a killer feature!

patrikaxelsson, 2020-05-08:
> Just an update: I have made a fair amount of progress on this, and with a few more hours of hacking I should have a useful subset of the API functions implemented.
>
> I was able to get amitcp_telnet [...]

All TCP/IP-stacks on the Amiga implementing the bsdsocket.library API requires kickstart 2.0+, so software written for it generally had no reason to support less - it will make it much easier for you to get software for testing if you use kickstart 2.0+ yourself.

niklasekstrom, 2020-05-08:
@patrikaxelsson I think you're right about the need for Workbench 2.0+.

Do you have some application in mind that uses bsdsocket.library? For me, downloading files and copying files on my network is sorted through "pi wget <file>" and a314fs.

jcagigal, 2020-07-14:
AmIRC would be nice. RPI could run BittleBee (https://www.bitlbee.org) and you could access Telegram, Discord and other interesting messaging applications.

perweij, 2026-10-10:
Hi Niklas, thanks for libremote. I've been looking at running a network server on the Amiga through it: [amiagent](https://github.com/thomas-luebker/amimcp), a small daemon that only needs `bsdsocket.library`. That needs `bind`/`listen`/`accept`, which are still stubs, so I'd like to fill in more of `bsdsocket.py`. I'll keep to the existing structure (`libdecl-bsdsocket.json` and the `handle_*_op` methods) so it can come back as a PR if you want it.

**The plan, in order:**

1. **One shared wait helper for blocking calls.** `library.c` already sends every signal to the Pi while a call is in progress, but only `connect` uses that today, in its own `select([sock, wait_sock])` loop. I'd move that into one helper, "wait until these sockets are ready, the timeout passes, or a signal in this mask arrives", and use it for `connect`, `accept`, `recv`, `send` and `WaitSelect`. `WaitSelect` would then honour its signal mask and return the signals that fired. Most Amiga network programs build their event loop on that. Non-blocking sockets (`IoctlSocket(FIONBIO)`) also become easy on top of it.
2. **BSD-to-Linux translation tables** for socket options and error codes. They're numbered differently on the two systems: `SOL_SOCKET` is 0xffff vs 1, `SO_REUSEADDR` 4 vs 2, `EAGAIN` 35 vs 11, `EADDRINUSE` 48 vs 98. Today most calls return -1 without setting errno.
3. **Shared helpers for reading and writing the 4.4BSD `sockaddr_in`** (the one with the length byte), for `bind`, `accept`, `getsockname`, `getpeername`, `sendto` and `recvfrom`.
4. **Then the server calls:** `bind`, `listen`, `accept`, `setsockopt`/`getsockopt`, `getsockname`, `getpeername`.

I'd test with wget first, since it already works, then amiagent.

**A design question I'd like your view on: which machine's ports should `bind()` use?**

The obvious answer is that `bind()` on the Amiga opens the same port on the Pi. That's very convenient: a client just connects to the Pi's address, and the Amiga needs no IP setup at all. Some consequences:

- a314d services run as a normal user, so ports below 1024 (FTP on 21, HTTP on 80) fail unless the service gets `CAP_NET_BIND_SERVICE` or the port is remapped.
- Amiga ports can clash with the Pi's own services, e.g. a telnetd or sshd on the Amiga.
- Binding to all addresses opens the Amiga service to the whole LAN. A real stack does the same, but here it's less obvious.

Options I can see:

- **(a)** Bind exactly what the Amiga asks for, and leave the rest to the user.
- **(b)** (a), plus an optional port-remap and allow-list setting, e.g. `21→2121`.
- **(c)** Bind only to localhost by default and open specific ports in config.

I lean towards (a) for now and (b) later if needed. Do you have a preference, or had you already planned something for the server side?

*(Disclosure: this analysis and plan were put together with help from Claude.)*

niklasekstrom, 2026-10-11:
Hello sir!

> Hi Niklas, thanks for libremote. I've been looking at running a network server on the Amiga through it: [amiagent](https://github.com/thomas-luebker/amimcp), a small daemon that only needs [...]

Yes please, if you end up with something useful then I'd be happy to merge it.

> **The plan, in order:**
>
> 1. **One shared wait helper for blocking calls.** `library.c` already sends every signal to the Pi while a call is in progress, but only `connect` uses that today, in its [...]

All of this sounds good, I think.

> **A design question I'd like your view on: which machine's ports should `bind()` use?**
>
> The obvious answer is that `bind()` on the Amiga opens the same port on the Pi. That's very convenient: a [...]

Yes same, I lean towards (a). I think I'd lean more towards `CAP_NET_BIND_SERVICE` than (b). However, if port remapping seems useful then it should be easy to add, so why not.

> _(Disclosure: this analysis and plan were put together with help from Claude.)_

Using more powerful tools to get more Amiga software seems like a good thing to me. Still good to disclose the usage for transparency! 👍