niklasekstrom/a314 #23: Component kits?

aiobofh, 2019-08-08:
Is there somewhere I can get already sourced komponent kits? I already have the PCB on the way, but just sourcing a few components (if you're not a die-hard electronics hoarder) is quite boring. :) I'm looking forward to be USING my A314 and start developing interesting things A.S.A.P. :) And not being used to sourcing, it feels both time consuming and quite pricy ordering just a few components here and there (or a shit-load of them that will just lay around).

A zip-loc with all the things needed for the build for sale? Someone? :) - I could also go for a already built one if anyhone have one extra.

nonarkitten, 2019-08-08:
https://octopart.com/bom-tool/1Z6N9nNS

aiobofh, 2019-08-08:
Brilliant! Thanks @nonarkitten :) You learn every day.

Eriond, 2019-08-08:
I made a DigiKey BOM a while ago that still should be valid: https://www.digikey.se/BOM/Create/CreateSharedBom?bomId=7822225
And to comment on _aiobofh_'s statement; yes, sourcing and building DIY hardware is not everybody's cup of tea. It is both time consuming and sometime pricey. That's just a decision you have to make: Either take on the challenge of building it yourself or wait around until someone else makes a batch that you could buy from.
It's not my intention to scare you away, but do mind that the A314 PCB is not a beginners project. I would place it somewhere 3-4 on a scale to 5. (1 being breadboard projects and 5 being BGA components) Considering myself a senior hardware developer, I have still made soldering mistakes/fails on 3 out of the 6 boards that I've built so far. And I had to use a really good microscope to find and correct two of them.
I sincerely wish you the best of luck, and hope that you will be able to enjoy our own A314 as soon as possible. If you get stuck, drop us a message on EAB forum, and we will do our best to help you out.

aiobofh, 2019-08-08:
@Eriond ordered the stuff :) Build time is comming. The BOM-import thing was quite convenient. Did not know about it :D

closed 2024-05-21 by niklasekstrom