niklasekstrom/a314 #24: Docker suggestion

terriblefire, 2019-08-15:
Hi, great project

You could reuse my dockerfile and docker code to give people build environments for the drivers... 

Take a look 

https://github.com/terriblefire/cdtv.device

terriblefire, 2019-08-15:
I just did this for you quickly.. here's a pull request.. doesnt touch any existing files 

https://github.com/niklasekstrom/a314/pull/25

niklasekstrom, 2019-08-15:
Hey Stephen, that's a great suggestion. I will try it out this evening. Thanks!

niklasekstrom, 2019-08-16:
That's excellent, thanks again.

I was planning to reorganize the directory structure a bit at some point, related to issue #15. If you have any thoughts or ideas about ways to do that I would be happy to hear them :)

closed 2019-08-16 by niklasekstrom