niklasekstrom/a314 #26: A2000 support?

amipieTOASTER2K, 2019-08-16:
Great work... 

  Amiga 2000 support should be fairly straight forward... Has a version for the A2000 commenced? Do I need to produce an adapter board of sorts for the A500 version to fit in an A2000? Unlike the A600, space shouldn't be an issue here. 

  I already have a raspberry PI mounted on a blank PCB in an unoccupied slot inside my A2000 case. A2000 supplying only power. Also using some KVM switches and sum A234 and more for I/O... Running AmiBerry... Need to have some Ami-Pie sex happening soon... Faster File transfers?, Perhaps shared storage and more... I do have an A500... I would rather install the A314 in the A2000... Currently sourcing components for build... Thanks guys...

niklasekstrom, 2019-08-16:
Hi amipieTOASTER2K,

Amiga 2000 support is actually not as straight forward as one might think. The A500 expects the board that is inserted in the trapdoor expansion slot to be DRAM chips and not much else, so the FPGA on the A314 must communicate using the same protocol as expected from DRAM chips. A board in an A2000 however communicates with the Amiga using the Zorro II protocol, which is quite different.

Moreover, memory that sits on a board in a Zorro II slot cannot be directly accessed by the custom chips (i.e. is not chip mem), which prevents some applications (VideoPlayer, PiAudio, RemoteWB) from working. There are, however, several applications that would still work well (a314fs, pi, bsdsocket.library), so I think a version for A2000 could still be interesting.

I don't know of any effort to build a Zorro II-version of A314. My feeling is that it shouldn't be all that hard, especially since the A500 version of A314 already exists. If someone were to design a Zorro II board I would certainly help out with adapting the A314 Verilog and software for it.

nonarkitten, 2019-08-16:
What you would want to do is make a small board that sockets into Agnus, this is where most of the signals from the Amiga 500 trapdoor come from as well as a few from Gary. This could be combined with a pin-shift to use the more common 8375 Agnus and would almost be a simple straight pin-out to the header.
```
DRD 15:0     AGNUS
DRA 8:0      AGNUS
_EXTICK      GARY
_CASL        AGNUS
_CASU        AGNUS
_RAS0        AGNUS
_RAS1        AGNUS
_WE          AGNUS
_CCK_A       AGNUS, Invert CCK*
_CLKRD       GARY
_CLKWR       GARY
D 3:0        AGNUS
A 5:2        AGNUS
```
* Note about CCK_A; the Amiga 500 inverts CCK to make CCK_A using a 74F04.

amipieTOASTER2K, 2019-08-17:
Hi Guys, 
  Thanks for the quick response. I will initially test an A314 on my A500... I was planning on using a non occupied Zorro-II slot (ultimately my current non-linked ami-pie slot). A Zorro board incorporating a header plug for your A314 board as well as a shielded ribbon cable to an Agnus socket module/board (daughter-board). I would hope to include the Megachip 2MB on the zorro card. Shielded clip (or solder) leads to Gary or A larger Agnus socket daughter-board incorporating a Gary via a through socket? (old wire wrap i.e long legs). 
  Yes, the Amiga's ability (or lack thereof) to drive signals as well as signal interference is a concern. I am willing to try though.... Out of the question with modern hardware. The Amiga is a little more forgiving at its frequencies.... Though I lack the experience here. A shielded ribbon cable might not be enough. Impedance mismatches, signalling/timing issues-skew/overshoot etc. 
  I have a modified Amiga 2000 utilizing a [Super-obese (EXTREMELY-FAT-LADY) Agnus 8372B 318069-03] he...he... ;-) and 2MB chip ram. 
Compatibilty issues with the Toaster also I guess... I would remove it first... Thanks for the heads up on the Hex inverter... which would live on the daughter-board with other buffering... 
  A2000 support could be a real can-o-worms... at least in my perceived form factor. Perhaps I should stop being a tight a$$ and just buy a Vampire... Anyone using a Vampire with a VideoToaster? I do love to tinker with my Amiga's... Cheers,

nonarkitten, 2019-08-19:
There should be no compatibility issues with Toaster with the A314 and we have users who use the Vampire and Lightwave all the time, so I'm assuming it works fine, but it's untested.

The Agnus/Gary adapter is the only solution for the A314 in the Amiga 2000. The Zorro slots will not work -- there is no try. It's a simple enough board -- maybe an afternoon of routing and adding one MOSFET to invert the CLK_A signal?

rdesaintleger, 2023-10-13:
Hi.

I'm quite new to Amiga Hardware addons. But according to schematics, the Amiga 2000 already have a clock chip on board. Isn't there a risk to have two clock chips sharing the same signals ?

A solution would be to use the _OVR signal to disable Gary address decoding process (but I have to admit that this way of doing things is not clear to me).

About Zorro implementation of the A314 it would be much more complicated. From my understanding of hardware and schematics, cards on the Zorro bus can master the 68k bus by issuing a request/acknowledge to buster. Once they do so, they can read/write anywhere including chip and fast as long as the target address is within the 24bits addressing of the original 68k bus (that may not be the case for the blizzard 2060 fast ram for example).

closed 2024-05-21 by niklasekstrom