niklasekstrom/a314 #28: AHI Audio via Raspi?

hgw77, 2019-10-15:
First of all, the a314 is a realy nice idea! 
To use the trapdoor expansion slot in that way could bring complete new possiblitys to the Amiga500 :-)

As I read about `piaudio` the first idea was, why it is not the other way? For Amiga500 there is no modern soundcard available so I was thinking it would be very nice to have a `ahi.device` or something to play amiga audio through the soundcard of the raspi :-)

aiobofh, 2019-10-15:
I am also in to trying to create an AHI-device with 14bit paula playback... Stereo mixer in the pi in other words.

Also a MHI-device for decoding mp3:s and such would be wonderful. 

I will try to get my card working tonight, if I am lucky

niklasekstrom, 2019-10-15:
Sounds like a great idea! 👍

Eriond, 2019-10-15:
@hgw77: Thank you for the kind words.
Before you do a driver to use the Raspberry Pi's audio output for the Amiga, you should consider the wide-spread criticism against the audio quality of Pi (https://hackaday.com/2018/07/13/behind-the-pin-how-the-raspberry-pi-gets-its-audio/ , the comments are especially interesting).
There is however a good reason for letting the Pi do the audio decoding, and that is if it was part of a RTG solution, where the HDMI would not only carry the video output from the Amiga, but the audio as well.
Even if the Amiga's "sound card" isn't modern, I would argue that it still produces audio with better quality compared to the output from the Raspberry's 3.5mm jack. But maybe it's just me being forgiving and nostalgic :) Perceived audio quality is such a personal experience.

hgw77, 2019-10-16:
@Eriond: thank you for your repsonse, and you are right the audio quality of the internal raspi is not that good but there are a wide range of USB-Soundcards with much better sound quality that you can use instead? :-)

By the way, I love the good old `Paula` and I am always impressed what this sound chip can do ;-) but on the other side I was always `dreaming` to have a better soundcard on my Amiga500 to play audio in 16bit quality.

aiobofh, 2019-10-16:
@hgw77 you can get as much as 14bit stereo at 56kHz (with more exotic screen modes) thats where AHI _mixing_ in the pi comes in, but then shuffle the mixed left/right as 14bit-hack sample data to Paula... it's an AWESOME upgrade without having to buy expensive audio hardware. :)

hgw77, 2019-10-16:
@aiobofh: yes thats what I mean with impressed. The `Paula` is more than 30 years old and can play audio in 14bit and 56kHz? Thats awesome and `C=` didn't know about that killer feature ;-)
But that feature comes at a cost, it needs a very powerfull CPU. With the installed Pi in the trapdoor maybe a standard Amiga500 (or with 68020/30) could play (with the help of the pi) an mp3 in 16bit quality ;-)

aiobofh, 2019-10-16:
@hgw77 yeah. But with a314 allowed to write chipmem the CPU usage (for playback) is close to zero om the Amiga side.

hgw77, 2019-10-16:
A now I understand! the PI is doing all the prepartions so the paula only needs to play :-)

On the other side if you have an PI-AHI Device you can use any player on amiga side (like Eagleplayer or AmigaAMP) to play audio on the PI, but than the amiga needs to handle the encoding of an mp3 or playing a mod file by itself and send the audio data to the PI.

aiobofh, 2019-10-16:
@hgw77 correct. And we could also have a MHI-device (the API used for stand-alone hardware mp3-decoders) too. just shuffle the mp3-data to the pi, let it decode it, and shuffle the 16bit wav back to the Amiga and then AmigaAmp or whatever shuffels it back to the PI using this AHI-device for 14bit stereo encoding. :) There could also be a short-cut raw-allocating paula directly with the MHI device, but it's not as system friendly, but would reduce the traffic between Amiga and Pi with that bidirectional raw 16bit stereo sample.

aiobofh, 2019-10-16:
... Hm. On second thought. I think MHI is supposed to hijack Paula directly, since it's constructed for stand-alon output devices. But I have not studied the interace enough to be 100% sure.

aiobofh, 2020-01-22:
OK, sadly newer AHI versions will probably not compile for a stock A500 with 68000. I guess if one have a 020-or-better accelerator the idea is still on. I will however not have such card in any near future :(

nonarkitten, 2020-01-23:
> OK, sadly newer AHI versions will probably not compile for a stock A500 with 68000. I guess if one have a 020-or-better accelerator the idea is still on. I will however not have such card in any near future :(

Have you tried [Run68017](http://aminet.net/package/util/misc/Run68017)? It implements many 68020 opcodes and might let your 68000 run AHI compiled for 68020.

aiobofh, 2020-01-24:
> Have you tried [Run68017](http://aminet.net/package/util/misc/Run68017)? It implements many 68020 opcodes and might let your 68000 run AHI compiled for 68020.

Nope, I have not tried this. Seems a bit of a flimsy solution. Besides... I can't really see why e.g. 6.0 should not work to compile for low-end 68k - at least not for just a simple PAULA mode (other drivers might not compile, but it's not really needed in this case since I've based my patch on the existing PAULA driver). I'll revisit this project... But then again. The Run68017 might be more useful to get any AHI-capable softwares (players) to actually run on 68000 tho. So it's stil a valuable tip. Thanks @nonarkitten :)