niklasekstrom/a314 #29: Program FPGA firmware from RPi

niklasekstrom, 2019-10-26:
It chould be possible to connect additional pins from the RPi pin header to the FPGA JTAG pins, and then use the RPi for programming the FPGA firmware.

The program described on this page has been used successfully to program the firmware: https://weekly-geekly.github.io/articles/343524/index.html

niklasekstrom, 2019-12-22:
The program mentioned above have been modified by its author (as suggested by Marcin) to make it configurable which RPi pins are used for the JTAG connection. https://translate.google.com/translate?hl=&sl=ru&tl=en&u=https%3A%2F%2Fmarsohod.org%2F397-pi4-m2rpi

peon374, 2020-04-20:
The program talked about looks like it is intended to be a JTAG debugging server.  If you just wanted to use the connection to be able to upgrade the firmware on the MAX10, I think openOCD should be able to do this.  If the build process used to make the POF goes one step further and makes a SVF file as part of your release, it could be pulled in and flashed without any Quartus toolchain.

Eriond, 2020-04-20:
That sounds really interesting! We would appreciate it if you could consider trying it out?

Eriond, 2020-04-23:
Issue resolved with updated PCB layout, v1.2

closed 2020-04-23 by Eriond