niklasekstrom/a314 #30: USB access via ANAIIS

nonarkitten, 2019-10-29:
This would grant access to the Pi's USB ports from ANAIIS, a USB stack for the Amiga which is compatible with the stock  7MHz 68000 CPU. They provide a libusb-based proxy driver for running on UAE which should serve as a good foundation. There is far less API here to implement than BSD sockets.

http://aminet.net/package/driver/other/anaiis_hostusb

aiobofh, 2019-10-31:
@nonarkitten yepp! Lovely! This is a good alternative for more powerful Amigas with A314, most def. For the purist this would be heavenly. Also Poseidon could be supported (since it seems more main-stream).

For low-end Amigas that want to benefit from the off-loading by the Pi other solutions are more appropriate - But the one does not rule out the other.

alexchaney, 2020-02-24:
Too much work.  Let Linux manage the mounting and unmounting of USB device natively and then just have a softlink to the pi0: directory.  Keep it super simple and let Linux just handle it all.

aiobofh, 2020-02-25:
@alexchaney the ANAIIS/Poseidon approach would enable more powerful A500/A600's to get ("any") USB device in native-world, not only storage devices. But I think the A314 makes most sense on an unaccelerated (or modestly accelerated) A500/A600 - hence doing the USB termination Linux-side and make special/minimal stuff to enable the Amiga-side to access it makes more sense, as you say.

But... It would still be a nice option to have the complete USB-stack Amiga-side, if someone is to implement #30 :)

neilDGD, 2023-12-20:
Been thinking of this, did anyone take this on and make a start and have anything to share? My thoughts are - PI side management of the usb itself (Os level or if need be another service), and hook into Poseidon on the Amiga side, but use the a314fs as a template because who wants to reproduce all that packet logic etc? Not sure i'm digging myself a hole into oblivion here...