niklasekstrom/a314 #38: The generated BOM is not in a format that works with suppliers

peon374, 2020-05-29:
Can we get an export of the BOM that works with octopart and/or digikey?

Eriond, 2020-05-29:
The BOM was never intended to be in a supplier-compatible format. It consists of 34 different components.
I believe that typing them in manually shouldn't stop anyone from building their own card. And, as a courtesy, I took the time to add DigiKey part numbers to make it even simpler. That's as much effort as I'm willing to spend on this matter.

closed 2020-05-29 by Eriond

peon374, 2020-05-29:
All you need to do is add a data format like CSV.  It is probably one of the default export options.

niklasekstrom, 2020-05-30:
@peon374 If you like to create such a CSV file then feel free to contribute the file back to the repo with a pull request. I don't think it makes sense that you insist that someone else should do that for you.

peon374, 2020-06-01:
Awesome, will do shortly.