niklasekstrom/a314 #39: Amiga 1200 version

w-A-L-L-e, 2020-06-08:
Really cool board, makes me want to swap my a1200 for an a500 (almost ;) ). I know the expansion port differs in the a1200 but would it theoretically be possible to have a similar expansion board for the a1200? This looks like a really cool way to get a modern linux combined with the old amiga I especially liked the cross compile and pi command line features that and the drive mounting and gfx/chip memory sharing gives endless possibilities. So the trick to share the chipmem ram is that possible on the a1200 also? Maybe we need to emulate a 68020 as well to allow this on a1200?

nonarkitten, 2020-06-08:
In theory yes. It would have to live on top of the clock port header and use pogo pins to hit the unpopulated header pads. There are two signals which I believe disable the on-board memory and could allow the board to take over. Ensuring good signal contact could be very challenging without a major motherboard modification (installing the missing headers).

w-A-L-L-e, 2020-06-08:
Cool, how do you know this ? I mean is there some good book or documentation you recommend? I have no problem in soldering some extra headers to my clockport and converting an a314 with jumper wires to an accelerator connector for my a1200 ;) (price is doable in belgium https://amigastore.eu/738-amiga-1200-accelerator-connector-for-the-edge-expansion.html). Also my trials might help you creating an a1200 version after this project is matured.

Eriond, 2020-06-08:
Hold your horses... Even though you'll probably get the signals from point A to point B correctly, there is the issue of "signal integrity", which will cause all kind of issues for you. I don't mean to rain on you parade, but if you rush into this without too fast, you will be disappointed.
This is an interesting idea though, and I will most certainly look into the matter.

nonarkitten, 2020-06-08:
To disable on-board chip RAM you need to bring /ROE high. This is pin 22 of P9A.

nonarkitten, 2020-06-08:
> Hold your horses... Even though you'll probably get the signals from point A to point B correctly, there is the issue of "signal integrity", which will cause all kind of issues for you. I don't mean to rain on you parade, but if you rush into this without too fast, you will be disappointed.

Yes, I did say that good signal contact would be the big challenge here.

niklasekstrom, 2020-06-08:
Hi @w-A-L-L-e!

The custom chip signals (those in the custom chip section in [this image](http://amigadev.elowar.com/read/ADCD_2.1/Hardware_Manual_guide/node02CB.gif)) are not available on the A1200 trapdoor expansion port unfortunately. The needed signals are available on the P9A/B ports however, but to get to those signals reliably you'd probably have to solder pin headers onto the motherboard (they are typically not populated, as seen in [this picture](https://bigbookofamigahardware.com/bboah/media/download_photos/a1200mb_rev1d4.jpg)).

Then there is the issue that the data bus to the chip ram is 32-bit on the A1200 but 16-bit on the A500. All pins on the MAX10 FPGA chip that is used on the current version of A314 are already taken, so another chip package would be needed (probably a BGA package). Moreover, the DRAM on the A1200 uses fast page mode, so the logic that arbitrates accesses between Amiga and Pi would have to be reworked completely probably.

So to your question if it is theoretically possible to build A314 for A1200; I guess so :) But in practice I think it would be a significant undertaking.

closed 2020-06-12 by niklasekstrom

w-A-L-L-e, 2020-06-18:
Thanks for the more detailed info. I was indeed hoping for something only described in first paragraph (just some extra jumper wires and a header to solder). However your explanation in second paragraph clears a lot of questions up for me. 

So basically a similar device could be made for the A1200 but we can't reuse the current A314 board. For A1200 we need a new pcb with larger fpga and quite some extra software development to make it possible. So maybe someday when you've exhausted customers from A500, A600, A1000 owners, hope you get around to the A1200 version of this type of expansion. 

It's a really cool idea using a modern SBC as expansion device and if it ever gets an A1200 version I'll be first in line to pre-order one. What really surprised me is that just a 4 wire spi connection is fast enough to do that bi-directional memory copying back and forth between pi and amiga 500.