niklasekstrom/a314 #42: Raspian lite doesn't come with some required python libs

peon374, 2020-07-28:
If you add 
```
sudo apt install python3-distutils python3-dev
```
to your build instructions or the build script it will make sure they are present and you won't get:
```
**bpls2gif.c:5:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.**
```
or 
```
cd bpls2gif ; python3 setup.py install
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    from distutils.core import setup, Extension
ModuleNotFoundError: No module named 'distutils.core'
```

niklasekstrom, 2020-07-28:
Sound good to me. I will add it to the wiki.

Regarding the change to the makefile, would you like to fork the repo, do the change, and then create a pull request?

I can of course do the change to the makefile myself, but I think it's better if you do the pull request because then it it is clear that it was contributed by you.

aiobofh, 2020-07-28:
Side-note: I completely stoped using raspbian in favor of ubuntu-server - i feel it's more stable when bumping release versions than raspbian. and more current packages, along with a richer repository.

peon374, 2020-07-28:
Done.  

Also I will give ubuntu-server a go with all the PI variants I have.  I assume that raspbian has a more comprehensive kernel but that is an assumption made because RPI foundation promotes Raspbian over other distros.

niklasekstrom, 2020-07-28:
Excellent, thanks!

closed 2020-07-28 by niklasekstrom