niklasekstrom/a314 #44: HowTo for flashing FPGA from Raspi on V1.2

MC68060, 2020-08-04:
Would be nice if there is a HowTo... I've never flashed a device from a Raspi.

Eriond, 2020-08-04:
I agree. We should add that to the wiki. 
Today it's a bit cumbersome, because it requires specialized software. A preferable solution would be if there was a utility, command-line or other, that could do the complete flashing procedure "in one go". AFAIK we don't have that. It would be a much appreciated contribution to the project, if someone was willing to take on such a task.

The MAX10 FPGA that we use in this project has a dual configuration feature, that allows to have one config "running" and another one in stand-by. The concept is that you could either flash the stand-by area and then reset, swapping the two, or use one of them as a "factory default" for a fall-back scenario (in case something doesn't work out with the primary config).
This does in theory allow for flashing from the Amiga side too, but of course requires quite some additional work on the HDL. More information about this feature can be found in [https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/max-10/ug_m10_config.pdf](https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/max-10/ug_m10_config.pdf)

niklasekstrom, 2023-01-05:
A wiki page about this was created here: https://github.com/niklasekstrom/a314/wiki/Program-FPGA-using-openocd-on-the-Pi

closed 2023-01-05 by niklasekstrom