niklasekstrom/a314 #60: a314fs crashes with yellow screen on A500+ with Kick 3.1 and TF536

gmcn42, 2020-12-09:
Hi everyone!

First off: great work, I'm really impressed by the whole project, especially the professional-looking hardware design!

Today I tried to get mine working, it's a Rev. 1.1 I bought off AmiBay.
My System: A500+, Kick 3.1, TF536 68030/50MHz, Indivision ECS V2

The A314 is jumpered to 2MB mode and Amiga Test Kit is reporting 2M Chip, memory test runs flawless. INT2 is connected.

After working out I had to explicitly enable the SPI bus on the Pi, I was able to use the pi command and videoplayer.

[image]
[image]

When I try to use a314fs, though, the Amiga crashes (Blinking LED of death, Yellow Screen).

[image]

`mount PI0:` seems to work fine but once I either `cd PI0:` or click `PI0:` in DirOpus, the crash occurs.

Any idea how I can narrow down what's wrong? I thought about removing the accelerator and trying it on plain 68000 but I'd have to scavenge the CPU from another machine.

niklasekstrom, 2020-12-10:
Hi @gmcn42,

First, well done with the troubleshooting and great to hear that everything is finally working well! We can add both A314+TF536 and a314eth+MiamiDX to the "tested and seems to work" list :)

Secondly, this is definitively a bug that should be fixed, so thanks for reporting it. That the Amiga crashes with a yellow screen is a bit surprising (and interesting!). I suppose it should be a simple fix though, to check in a314fs.py if the shared directory exists, and if not then perhaps create it, or print a helpful error message (and certainly not crash).

A related thing that has been talked about, but no one got around to (yet), is to write a small program that installs the software needed on the Pi, and verifies that all files and configurations are as they should.

cnilsson, 2021-08-13:
I've reproduced this using kickstart 2.0 (albeit without the yellow screen - guru "only").

There is no error handling when there is no a314shared directory leading to a314fs.py throwing an exception and bailing out and thus stops responding to requests. The easiest way to fix this is probably to have a314fs.py create the directory as suggested. It looks like @niklasekstrom already had plans to handle problems such as a non responsive a314fs in a314fs.c but understandably moved on since everything worked.

niklasekstrom, 2023-08-13:
An installer script has been added, that amongst other things creates an a314shared directory.

Possibly/probably the code for a314fs should handle that the a314shared directory is missing, but I'll close this issue for now in any case.

closed 2023-08-13 by niklasekstrom