niklasekstrom/a314 #62: PCB standoff holes for securing Raspberry PI

prgp, 2020-12-24:
I noticed that the image [a314_with_rpi.jpg](https://github.com/niklasekstrom/a314/blob/bc3836feaac5206eead7e3241f3c7c8cd64175b7/Documentation/Images/a314_with_rpi.jpg) shows a Raspberry Pi (model unknown) dangling from the GPIO pins - modern Raspberry PI (later revision 1 plus models and later IIRC) have common hole placements for standoffs, would it be possible to incorporate at least one if not two of the standoff holes on the top of the board so that the Raspberry Pi can be secured properly to the board (as one would a normal Raspberry HAT).

Likewise on both the A500 and A600 boards having at least one standoff hole that conforms with a Raspberry Pi Zero would also be most helpful.

Eriond, 2020-12-24:
The RPi in question on that picture is the Model 3. Yes, you're right about it not having any stand-offs, but that is the result of numerous iterations and thorough tests of the PCB and it's mechanical integrity. You see, stand-offs are intentionally excluded for several reasons:
1) Excluding them allow for a a better and more relaxed routing of the PCB traces, avoiding us to use unnecessary thin traces or close track spacing.
2) The A314 system (meaning the A314 board together with the RPi attached on top, inside an Amiga) is in a physically secure location in space, not prone to vibration or chock. The RPi is secured by the 40-pin header as well as being fixed between the A314 PCB and the Amiga keyboard backplate. The connector isolator which you see in the picture you linked to (the white TPU strip on top of the RPi GPIO connector) makes contact with said backplate. There is just no room for lateral movement.
3) Further, the RPi lays with its video and audio connectors resting on top of the trapdoor side ledge, providing excellent support for the board. The only situation where this could be a possible issue, is if you were to use your Amiga upside-down. I'd say that is a an edge-case scenario...
I hope this eases your worries about the mechanical aspects of the A314 construction.

prgp, 2020-12-24:
I just saw the reverse of the board on Ebay and noticed that one of the distance standoff holes would have in the same location as the CP2032 battery holder, so that one is definitely not practical :-D

I assume that whilst a Raspberry Pi Zero would not suffer from the same level mechanical stress as a full sized model, but it would also not benefit from the video and audio connectors positioning - looking at the current design it would also put a hole rather close to IC4 (SRAM) and that is impractical to move.

What are the expected height specifications for the GPIO socket and pin headers (I assume that the pins are assumed to add little to no separation, so something like [this header](https://thepihut.com/products/colour-coded-gpio-headers) would cause issues) ?

Eriond, 2020-12-24:
We have based the design upon the RPi 3 as it is supplied by the RPF. For that model (and compatible ones), you will find the appropriate header component listed in the BOM.

closed 2024-05-21 by niklasekstrom