niklasekstrom/a314 #63: Raspberry Pi Compatibility

prgp, 2020-12-24:
I checked the wiki and was not able to locate a Raspberry Pi compatibility list.

As I understand it the Raspberry Pi is drawing power via the A500's trapdoor - what are the draw limits that this imposes on the Rapsberry Pi ? (a Zero has a maximum draw of about 300 mA and a Model 2B has a maximum draw of about 820 mA, but a model 4B has a maximum draw of about 1250 mA)

Would it be possible to add to the wiki which Raspberry Pi models have been used successfully.

alestanga, 2020-12-30:
I'm using a RPi 2B in my Amiga 500 and it works flawlessly, well, if you don't mind the absence of WiFi.
ethernet works just fine though.

Eriond, 2020-12-30:
compatibility list added. Issue may be closed.

closed 2021-02-09 by niklasekstrom