niklasekstrom/a314 #64: Automatic Pi Shutdown

prgp, 2020-12-24:
The wiki suggests that before you power off the Amiga that

> Before powering off your Amiga you should do `pi sudo shutdown now` and wait a few seconds to let the Raspberry shutdown properly.

Would it be possible to have power loss on the Amiga trigger a safe shutdown on the Raspberry Pi ?

I have seen Raspberry Pi projects that have used 'soft power' buttons connected to he GPIO to trigger a safe shutdown, but there would still need to be power to facilitate the shutdown (other projects such as [this](https://www.sensorsiot.org/safe-shutdown-for-raspberry-pi-with-super-capacitors-133/) appear to provide this, but the capacitors seem a little large for that purpose - 2A @ 5V for 5s should need only about 2F unless my maths is way off and much less than that for a Zero).

This could be implemented as an optional 'sister-board` to the Raspberry Pi thus allowing the rating (or inclusion) to be determined by the person attaching the Raspberry Pi.  The board probably would not need that many pins - Vin and Vout, GND and the control ('shutdown') line.

Eriond, 2020-12-24:
I suggest this is linked with issue #16

prgp, 2020-12-25:
I looked at the open and closed issues to see if it had been suggested, some how completely missing #16 ...

The only issue that I could see and could not come up with a simple solution for was if the PSU power-cycles too quickly then the shutdown could complete after power has been restored which means that the Raspberry Pi would not turn on again.

Maybe some sort of timer / latch that begins on power loss and should not complete unless power is returned too soon then interrupt the power, but this feels kludgey to me - the grounding RUN (via resistor for safety) reset method seems to rely on pogo pins or modifying the Raspberry Pi.  Also the Zero,  model 3 and model 4 do no have RUN in the same location so pogos are probably doubly out.

closed 2020-12-25 by prgp