niklasekstrom/a314 #94: `Unable to connect to picmd service` / `Unable to open a314.device` on A1200 with clockport if

gmcn42, 2024-01-18:
Hi there,

I am trying to set up the clockport version of A314 with no luck so far. I got the hardware from a nice person on Discord who seemed to know what he was doing and tested it thoroughly before sending it to me, so here's hoping it's not a hw fault.

# What I did

1. Installed everything in the Amiga, the Pi with a pre-setup SD card
2. Boot amiga, connect to raspi via ssh
3. follow raspi-side installation instructions from latest `main` and reboot. systemd reports:
```
sudo service a314d status
● a314d.service - A314 Daemon
     Loaded: loaded (/lib/systemd/system/a314d.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2024-01-18 23:30:40 CET; 3min 14s ago
    Process: 504 ExecStartPre=/opt/a314/start_gpclk (code=exited, status=0/SUCCESS)
   Main PID: 513 (a314d)
      Tasks: 1 (limit: 421)
        CPU: 41ms
     CGroup: /system.slice/a314d.service
             └─513 /opt/a314/a314d

Jan 18 23:30:40 a314-1200 systemd[1]: Starting A314 Daemon...
Jan 18 23:30:40 a314-1200 start_gpclk[504]: Starting GPCLK0 with frequency=85 (requested=90), divisor=14
Jan 18 23:30:40 a314-1200 systemd[1]: Started A314 Daemon
```
4. copied over the Amiga binaries using ADF opus and FlashFloppy
5.  Copied everything from virtual floppy to hard disk using DOpus 4
6.  rename a314-cp.device to a314.device, delete -fe and -td devices
7. Try out `pi` command
```
Unable to connect to picmd service
```

# What I already tried

1. Re-flashing the CPLD using openFPGALoader and the v1.1 firmware blob available in the clockport_interface repo
    - openFPGALoader takes a long time to flash (minutes) but reports success
    - a314d is fine after reboot
    - still same error
2. Using a314d from 1.2.2 tag instead of main to match amiga binary versions
    - now I am getting a few seconds of wait after typing `pi`, and then `Unable to open a314.device`
3. reseating the module on the clockport header multiple times
    - no effect

# Ideas

* While the Pi side seems to have a good connection to the cpld, I am unsure about the Amiga side. Is there any way to get more verbose output and/or check if the CPLD is present without actually trying to connect to the Pi?

gmcn42, 2024-01-19:
Additional info: For flashing I used the instructions at https://github.com/katzazakis/A314cp-SD-brew/wiki/How-to-flash-your-A314cp with the JED file https://github.com/niklasekstrom/clockport_pi_interface/releases/download/v1.1/cp_pi_if.jed

The board is the original design, not the eriond one and carrying a Pi 3A+

niklasekstrom, 2024-01-19:
Hi,

If you are following instructions other than the ones given in this repo (or the clockport_pi_interface repo) then you need to ask whoever wrote those instructions if they are not working.

In general, if you are buying a board that someone else built, then the person who built that board should flash it and test it, and you should not need to flash it again. Please don't flash a working board.

closed 2024-01-19 by niklasekstrom

gmcn42, 2024-01-19:
Alright, two remaining questions then:

1. For debugging: Does `Unable to connect to picmd service` imply the CPLD was reachable but not the Pi? Does `Unable to open a314.device` mean it can't reach the CPLD?

2. Regarding flashing instructions: Are there official ones? In the clockport-interface repo I am only seeing build instructions but nothing related to flashing the CPLD

niklasekstrom, 2024-01-19:
2. If you didn't build the board then you shouldn't have to flash the cpld. If you are building boards then likely you can figure out how to flash the cpld, or you can come to the Discord server to ask how to do it.

1. Unable to open a314.device typically means that the Amiga side is unable to communicate with the cpld. This can happen for several reasons, such as software configuration/versioning issues.

Feel free to ask more questions in the Discord server.