niklasekstrom/a314 #103: Failed pi install

SanguineBrah, 2024-12-07:
I had a working system but messed up my pi install and had to rebuild again, so I followed the instructions in the software directory again. I'm using the trapdoor variant.

The pi command works on the Amiga side but the SANA driver no longer works. This is the log from systemctl (a314d)

Dec 07 17:29:51 a314 systemd[1]: Started a314d.service - A314 Daemon.
Dec 07 17:29:52 a314 a314d[541]: ERROR, 2024-12-07 17:29:52,646, __main__, line 218: Unable to open tap device at tap0
Dec 07 17:29:52 a314 a314d[541]: Traceback (most recent call last):
Dec 07 17:29:52 a314 a314d[541]:   File "/opt/a314/ethernet.py", line 263, in <module>
Dec 07 17:29:52 a314 a314d[541]:     tap.close()
Dec 07 17:29:52 a314 a314d[541]:     ^^^
Dec 07 17:29:52 a314 a314d[541]: NameError: name 'tap' is not defined. Did you mean: 'map'?

ifconfig does not list any bridge device

This is the output from the install script:

g++ -DMODEL_TD=1 a314d/a314d.cc -O3 -o bin_pi/a314d-td
dtc -I dts -O dtb -o bin_pi/spi-a314.dtbo a314d/spi-a314-overlay.dts
created virtual environment CPython3.11.2.final.0-64 in 1570ms
  creator CPython3Posix(dest=/opt/a314/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./bpls2gif
  Preparing metadata (setup.py) ... done
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8289c9d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pyudev/
Collecting pyudev
  Downloading https://www.piwheels.org/simple/pyudev/pyudev-0.24.3-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 67.9 kB/s eta 0:00:00
Collecting websockets
  Downloading websockets-14.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (169 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 169.4/169.4 kB 937.9 kB/s eta 0:00:00
Collecting python-pytun
  Downloading python-pytun-2.4.1.tar.gz (8.5 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: python-pytun, bpls2gif
  Building wheel for python-pytun (setup.py) ... done
  Created wheel for python-pytun: filename=python_pytun-2.4.1-cp311-cp311-linux_aarch64.whl size=28170 sha256=f464f5e52f9bfd8bc69dbb53358c1d3ebc55e5ae5d0a42838aa9b2f43c9fe280
  Stored in directory: /root/.cache/pip/wheels/96/05/30/49ad90b4a08cc1fd6a875874f2c3f11b1650ae00ed279bec7b
  Building wheel for bpls2gif (setup.py) ... done
  Created wheel for bpls2gif: filename=bpls2gif-1.0-cp311-cp311-linux_aarch64.whl size=14423 sha256=ad20d7f726fd1cd9727d2b2126be862a66ae036d93bd565c9917617ea30727e2
  Stored in directory: /tmp/pip-ephem-wheel-cache-yv0ws32i/wheels/3f/57/8a/5111eee2bf6156a17289951e2ecf8e60a4b4678cf303d0402c
Successfully built python-pytun bpls2gif
Installing collected packages: python-pytun, bpls2gif, websockets, pyudev
Successfully installed bpls2gif-1.0 python-pytun-2.4.1 pyudev-0.24.3 websockets-14.1
./install-pi.sh: line 15: /etc/network/interfaces.d/tap0: No such file or directory
Created symlink /etc/systemd/system/multi-user.target.wants/a314d.service → /lib/systemd/system/a314d.service.
Created symlink /etc/systemd/system/multi-user.target.wants/a314net.service → /lib/systemd/system/a314net.service.

Installation complete
Restart the Raspberry Pi (sudo reboot now) to automatically start a314 software

niklasekstrom, 2024-12-08:
Hi, can you try running the script again? It looks in the output like the network connection was broken at some point, so something was likely not downloaded properly.

SanguineBrah, 2024-12-08:
I just redid everything from scratch - reimaged the SD card using raspberry pi imager to Lite 64bit then followed the instructions exactly. This is the output from the script:

g++ -DMODEL_TD=1 a314d/a314d.cc -O3 -o bin_pi/a314d-td
dtc -I dts -O dtb -o bin_pi/spi-a314.dtbo a314d/spi-a314-overlay.dts
created virtual environment CPython3.11.2.final.0-64 in 1567ms
  creator CPython3Posix(dest=/opt/a314/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./bpls2gif
  Preparing metadata (setup.py) ... done
Collecting pyudev
  Downloading https://www.piwheels.org/simple/pyudev/pyudev-0.24.3-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 813.9 kB/s eta 0:00:00
Collecting websockets
  Downloading websockets-14.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (169 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 169.4/169.4 kB 1.0 MB/s eta 0:00:00
Collecting python-pytun
  Downloading python-pytun-2.4.1.tar.gz (8.5 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: python-pytun, bpls2gif
  Building wheel for python-pytun (setup.py) ... done
  Created wheel for python-pytun: filename=python_pytun-2.4.1-cp311-cp311-linux_aarch64.whl size=28169 sha256=e1e687cdfea72a4f2700c0ba30bd2b5158a377ae472413569453d177e58354e0
  Stored in directory: /root/.cache/pip/wheels/96/05/30/49ad90b4a08cc1fd6a875874f2c3f11b1650ae00ed279bec7b
  Building wheel for bpls2gif (setup.py) ... done
  Created wheel for bpls2gif: filename=bpls2gif-1.0-cp311-cp311-linux_aarch64.whl size=14423 sha256=ed922da99e6e219d2fbb2c14e0b1362e6cdf9cd16388f99a32e517a63485fd18
  Stored in directory: /tmp/pip-ephem-wheel-cache-umiecv69/wheels/3f/57/8a/5111eee2bf6156a17289951e2ecf8e60a4b4678cf303d0402c
Successfully built python-pytun bpls2gif
Installing collected packages: python-pytun, bpls2gif, websockets, pyudev
Successfully installed bpls2gif-1.0 python-pytun-2.4.1 pyudev-0.24.3 websockets-14.1
./install-pi.sh: line 15: /etc/network/interfaces.d/tap0: No such file or directory
Created symlink /etc/systemd/system/multi-user.target.wants/a314d.service → /lib/systemd/system/a314d.service.
Created symlink /etc/systemd/system/multi-user.target.wants/a314net.service → /lib/systemd/system/a314net.service.

Installation complete
Restart the Raspberry Pi (sudo reboot now) to automatically start a314 software

The SANA driver still fails; it seems like the line to add the tap0 inferface didn't do anything. There is no inferfaces.d in /etc/network

niklasekstrom, 2024-12-08:
In https://downloads.raspberrypi.com/raspios_lite_armhf/release_notes.txt it says " * Removed ifupdown from default install as it is not used and slows down boot", maybe this is the problem.

If you do `sudo apt install ifupdown`, and then run install-pi.sh again, does that help?

If you have a Discord account you could join the A314 discord server (link on https://github.com/niklasekstrom/a314?tab=readme-ov-file#do-you-want-to-get-involved) to get help more interactively.

SanguineBrah, 2024-12-08:
Installing ifupdown did the trick. Looks like it just needs adding to the dependencies in the install instructions.

niklasekstrom, 2024-12-08:
Ok great, thanks for reporting the issue!

If you know of some better way to configure the tap interface (than the current method) then feel free to drop by the discord server and let us know. Otherwise I'll add the ifupdown package to the dependencies in the install instructions.

closed 2024-12-08 by niklasekstrom