niklasekstrom/a314 #104: next adaptor

alow2004, 2025-02-13:
Hello,
Would you agree to develop such a solution for the next computer ? It's the same processor family as the Amiga (68040), but using nextstep as operating system. A think it could be feasible to adapt the 'amiga' part for another 68K computer. If you agreed, I could send you a next station so you could develop the prototype ?
What is your advice about that? Thanks for your response. And sorry using the issue channel, but it was the only way I found to contact you. best regards.

cnilsson, 2025-02-13:
What kind of functionality would you like to add to NextStep? Sure there may be  some niche use cases, but I am having a hard time seeing what features you are looking for, at least among the most commonly used a314 ones.

It is a unix system after all so if you want a shell on a Pi you are better off using ssh. You also already have access to file sharing such as NFS or SMB over ethernet. Storage, well I assume you need a disk (or SCSI emulator) anyway for the OS so no need for that. Audio - possibly better DAC on the Next than a Pi. Well it won't do Spotify if that is what you are after but you can ssh to another machine and get Spotify with GUI and everything over X.

Also, check the discord link for a better way to discuss with Niklas and others.

alow2004, 2025-02-13:
Hello, thank you for your response,
Well, first of all, we don’t have direct wi-fi functionalities available for Next, the nextbus, that is an avatar of Mac’s nubus is not so easy to interface because next used proprietary asics to manage
The protocol, and those ic’s are now rare to find, so I was thinking of a more simplified interface based on the 314-cp (clockport), after all, it’s only 8 bits data bus, some address lines, rw signals and one interrupt…
There is a signal processor available on the Next (DSP 56001) that could lead also to ’funny’ usage. The fs utilities because actually, next is based on scsi disk and let’s say it frankly, it could be very handy
To have the Raspberry pi totally integrated into the next box in place of juggling with external links, external computers etc…
By example, I also follow-up the pistorm project, actually, it’s not really useful because next are natively equipped by 68040 processors, but it’s only at 25 MHz, there is no accelerator available anymore, so I bet that
If the pistorm project continue its evolution (maybe using RPI5 ?), it would become a very attractive solution for next community if an adaptor is available in 68040 socket format... 

Have anice day!

niklasekstrom, 2025-02-15:
Hi @alow2004,

Does the next station has some connector that would be suitable to connect it to a Raspberry Pi, possibly with a cpld/fpga in between?

alow2004, 2025-02-16:
Well, there is the NextBus, as said, it is the same as a Mac Nubus, but supporting twice the speed (20 MHz), the protocol is nubus Texas instrument based. But other than that, the only way I see is 
Pluggin into the processor socket or the high speed serial links of the 56002 DSP processor

You can find the nextbus specs in attachment.

https://www.nextcomputers.org/NeXTfiles/Docs/Hardware/NeXTbus-NBIC/NeXTbus_Specification.pdf?__cf_chl_tk=2zvCxOUliJxolNsHWt6bU.qsmiP6XcuZRYbv5WL68NM-1739664130-1.0.1.1-42x6WrmMhoyTIWouDKl.wDh7qHJe0FZZ4YnRM8bQDlI

> Le 15 févr. 2025 à 23:51, Niklas Ekström ***@***.***> a écrit :
> 
> Does
￼

niklasekstrom, 2025-02-16:
Building a hardware adapter that sits between the NextBus and a Raspberry Pi is likely possible, but it would be a significant effort. It requires designing and building hardware, writing and debugging FPGA logic, and writing and debugging software. As I don't have a NextStation (and never had), I don't have sufficient incentives to spend that effort. 

One much simpler possibility might be if you connect the NextStation and the Pi together through a serial cable, and run some software (PPP?) that lets the NextStation use the Pi's network interface.

alow2004, 2025-02-16:
> niklasekstrom

Yes, you advice is valuable in that sense, I think what I will do is also checking on the MAC community because as is said, NextBus is a direct ’successor’ of the Nubus standard for Mac,
And there were a lot of accelerators for the Mac at the time, I will try to check how they handle the protocols and see if there are opportunities.
Thanks for you help and advices, have a nice day!

closed 2025-02-17 by niklasekstrom